MajicDesigns/MD_Parola #97: Issue of using the library on a Arduino UNO Wifi Rev 2...

Wettemark, 2022-01-30:
I'm developing a scoreboard for a baseball club in the local town. For this I have used the Arduino UNO WiFi Rev 2 card as I'm going to connect the control and the board through WiFi.

I'm using the MD_Parola library verison 3.5.6 to control the matrixes. I'm using 4 matrixes split up into 4 zones to display one value for each of the zones.

I did a lot of the development on a normal UNO and everything worked good and still does when I'm running the code without WiFi parts. But as I shifted over to the UNO Wifi card I started to get problems.

Both the Wifi and the Parola library are using SPI interface. I have needed to shift the CS for the Parola library to a different pin. I'm running it over pins 11, 12 and 13.

But I still have issues. Right now I can get 3 of the four zones to work, but the last zone will not display what I send to it. Right now it's showing all elemets lighted up and quite bright. Does not matter what I push out to the display, will not show.

I have tried a couple of different pin setups but as soon as I run the code on the UNO Wifi it will not light up the 4th zone. All the other three works all the time.

I have a test program that works on a normal UNO but as soon as I upload this to a UNO Wifi this happens.

I have check the hardware and and all matrixes are working.
Wireing is ok, using the same just shifting the feeding connector between them.

I honestly do not know what else to test. Please help!

Snippet of code:
#define CS_PIN          12     //Pin for SPI interface - CS
#define DIN_PIN         11    //Pin for SPI interface - DIN
#define CLK_PIN         13    //Pin for SPI interface - CLK
  
// Configuration for matrixes
#define HARDWARE_TYPE   MD_MAX72XX::GENERIC_HW
#define MAX_DEVICES     4

// Create a new instance of the matrix controller and also using SPI Interface
MD_Parola matrix = MD_Parola(HARDWARE_TYPE, DIN_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

viod setup(){
  // Intialize the led matrix and setting up the 4 zones, one per zone
  matrix.begin(4);
  matrix.setZone(0, 0, 0);
  matrix.setZone(1, 1, 1);
  matrix.setZone(2, 2, 2);
  matrix.setZone(3, 3, 3);

  
  // Set the intensity (brightness) of the display (0-15)
  matrix.setIntensity(brightness-1);

  // Clear the display
  matrix.displayClear();

  //Setting up the zones on the led matrix
  //matrix.setFont(_seg7_fixed_medium);
  matrix.setFont(_seg7_fixed_small);
  matrix.setCharSpacing(0); //Set for all zones
  matrix.setZoneEffect(0, true, PA_FLIP_UD);
  matrix.setZoneEffect(0, true, PA_FLIP_LR);
  matrix.setCharSpacing(1, 1);
  matrix.setZoneEffect(1, true, PA_FLIP_UD);
  matrix.setZoneEffect(1, true, PA_FLIP_LR);
  matrix.setCharSpacing(2, 0); //Set special handling for zone 2
  matrix.setZoneEffect(2, true, PA_FLIP_UD);
  matrix.setZoneEffect(2, true, PA_FLIP_LR);
  matrix.setCharSpacing(3, 1);
  matrix.setZoneEffect(3, true, PA_FLIP_UD);
  matrix.setZoneEffect(3, true, PA_FLIP_LR);
  
  //Connect the zones to char buffer 
  matrix.displayZoneText(0, szInnings, PA_LEFT, 0, 0,  PA_PRINT, PA_NO_EFFECT  ); //2
  matrix.displayZoneText(1, szHome, PA_LEFT, 0, 0,  PA_PRINT, PA_NO_EFFECT  ); //0
  matrix.displayZoneText(2, szAway, PA_LEFT, 0, 0,  PA_PRINT, PA_NO_EFFECT  ); //1
  matrix.displayZoneText(3, szBrightness, PA_LEFT, 0, 0,  PA_PRINT, PA_NO_EFFECT  ); //3
}

void loop() {
  sprintf(szInnings, ";%d", innings); 
  sprintf(szHome, "%d", homeScore);  
  sprintf(szAway, "%d", awayScore);  
  sprintf(szBrightness, "%d", brightness);  

  //Updating the matrixes
  matrix.displayAnimate();
  matrix.displayReset(0);
  matrix.displayReset(1);
  matrix.displayReset(2);
  matrix.displayReset(3);
}

MajicDesigns, 2022-01-30:
Snippets don't work too well because I cannot see what else your program is doing, which is often where the issues are.

As I understand your requirements: you have 4 matrices; each matrix is a zone; each matrix will display a single numeric digit; the only animation used is PA_PRINT.

From what you have given, I think the structure of your loop() may be causing a problem because you are continuously sending the data (hundreds of times a second) and erasing the data immediately with the displayReset(). This may be just in the smippet , but that is what I can see.

PA_PRINT will take one animation cycle to display, so I imagine this is a lot faster than your baseball score will change. I would make sure the loop only updates the display when the data changes, more like
`
matrix.displayAnimate()
if (any of the data has changed)
{
  sprintf(szInnings, ";%d", innings);
  sprintf(szHome, "%d", homeScore);
  sprintf(szAway, "%d", awayScore);
  sprintf(szBrightness, "%d", brightness);
  matrix.displayReset(0);
  matrix.displayReset(1);
  matrix.displayReset(2);
  matrix.displayReset(3);
}
`
You keep track of when the data changes either by setting an 'old' value for each and then comparing to the current or more easily (with such a small amount of data) by just setting a global boolean flag when you change any of the displayed values. This is then used to trigger the update and reset to false in the 'if' statement.

An alternative is to get rid of the zones altogether. You are using a fixed font, so the spacing will be constant ant you can work out where things will land on the display easily. Then use
`
sprintf("%d %d %d %d", innings, homeScore, awayScore, brightness);
`
to format one string for the display. Using one zone you can just use the .print() method or the native displayAnimate() method (however, the same comments as above apply for updates in loop()).

Wettemark, 2022-01-31:
Thanks for your reply. 

I'm not to used to writing codce for the Arduino and I see your point about updating the matrixes to often. There is a bit of code where I today are updating the values to show in the matrixes so I will move the displayReset command to inside the if statements for the update. I will report the progress.

I would just lift one question, the code is working correctly on a normal Arduino UNO but it's only on a Uno Wifi I'm getting this issue. I can reconnect to a UNO and use the same hardware for the matrixes and buttons and I get it to work with the same test program. I understand that I got a collision on the UNO Wifi since the network also using the SPI interface so that I have fixed. But the last of the displays will not work on the UNO Wifi.

Are there some differences in the architecute between the two more than that it's a built in WiFi shield on the UNO Wifi?

Wettemark, 2022-01-31:
Sorry, not used to Githud... ;-)

Wettemark, 2022-01-31:
Main code of the program

`/*****************************************************************
 * Program to han dle the score controling os a basevall game. The score and information will 
 * be sent to the ScoreBoard over IP.
 * It's only possible to have one ScoreControl but it should be possible to have multiple ScoreBoards
 * 
 * Format of input information are:
 * Message format
 * 1111111111222
 * 1234567890123456789012
 * #I=XXH=XXA=XXO=XB=XXX&
 * I=XX - Inning with first X to include : for TOP and ; for BOTTOM, Second X tgo include the inning number
 * H=XX - Home score with zero filled, one of two digits
 * A=XX - Away score with zero filled, one of two digits
 * O=X - Outs, will be 0, 1, 2 or 3
 * B=XX - Brightness with zero filled, one of thress digits (0-255)
 * & - end line sign 
 *
 * Program controlls the following buttons and matrixes:
 * # BUTTONS
 * # Inning count up and down. (Low value is TOP 1 and high value is 0 BOTTOM which is the extra inning)
 * # Home score up and down 
 * # Away score up and down
 * # Outs (Cycles from 0 to 3 and then resets to 0)
 * # Brightness (To control the brightness of the lep strips in the ScoreBoard)
 * # Reset (NEEDS TO BE DOUBLE CLICK TO RESET)
 * # MATRIXES
 * # Innings (Showing both arrow for TOP and BOTTOM and innings number)
 * # Home Score (Max score is 29)
 * # Away Score (Max score is 29)
 * # Brightness for led strips (1 - 16) (When communicated to ScoreBoard it will be timed by 17 to move value to 0 - 255)
 * 
 * Sample data
 * #I=:1H= 0A= 0O=0B=032¤ (TOP first inning, Home 0, Away 0, Outs 0, Brightness 68)
 * #I=;3H= 7A= 3O=2B=032¤ (BOTTOM third inning, Home 7, Away 3, Outs 2, Brightness 68)
 * #I=:5H= 9A= 4O=1B=032¤ (TOP fifth inning, Home 9, Away 4, Outs 1, Brightness 68)
 * #I=;9H=16A=10O=3B=032¤ (BOTTOM ninth inning, Home 16, Away 10, Outs 3, Brightness 68) (Mapped from 4 in ScoreControl)
 *
 *****************************************************************/

#include <Bounce2.h>
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "score.h"
#include "seg7_fixed_medium.h"
#include "seg7_fixed_small.h"

//Definition of pins for input and output
#define INNINGDOWN      2     //BUTTON
#define INNINGUP        3     //BUTTON
#define HOMEDOWN        4     //BUTTON
#define HOMEUP          5     //BUTTON
#define AWAYDOWN        6     //BUTTON
#define AWAYUP          7     //BUTTON
#define OUTS            8     //BUTTON
#define BRIGHTNESSDOWN  A5     //BUTTON
#define BRIGHTNESSUP    A4     //BUTTON
#define RESET           A3     //BUTTON
#define OUT1            A2    //LED 
#define OUT2            A1    //LED 
#define OUT3            A0    //LED 
#define CS_PIN          12     //Pin for SPI interface - CS
#define DIN_PIN         11    //Pin for SPI interface - DIN
#define CLK_PIN         13    //Pin for SPI interface - CLK
  
// Configuration for matrixes
#define HARDWARE_TYPE   MD_MAX72XX::GENERIC_HW
#define MAX_DEVICES     4
[... 450 lines cut ...]
//Function to handle the outs diods
void lightOuts(int o) {
    //Setting the number of outs and lighting the leds
    switch (o) {
      //Clearing outs
      case 0:
        digitalWrite(OUT1, LOW);
        digitalWrite(OUT2, LOW);
        digitalWrite(OUT3, LOW);
        break;
      //First out
      case 1:
        digitalWrite(OUT1, HIGH);
        break;
      //Second out
      case 2:
        digitalWrite(OUT2, HIGH);
        break;
      //Third out
      case 3:
        digitalWrite(OUT3, HIGH);
        break;           
    };
}
`

Wettemark, 2022-01-31:
I have continued testing and on a Arduino UNO it works with my orignial program, the rewritten one and with a one where I tried to remove the zones and just use the displayText instead and not having zones. 

But on the Arduino UNO WiFi Rev 2 card it will never display the right information in the last matrix of the chain...

MajicDesigns, 2022-02-01:
> But on the Arduino UNO WiFi Rev 2 card it will never display the right information in the last matrix of the chain
Just to be clear , do you mean matrix number 0 (first one nearest the arduino connection) or number 3 (further from the Arduino connection).

Wettemark, 2022-02-01:
The matrix number 3.

MajicDesigns, 2022-02-02:
I really can't see what could be causing this. BTW, I see no 'Wifi' code in the code you posted, just straight out code to handle switches and the display.?

Have you tried one of the examples from the Parola library on the Uno Wifi board? Does it manage to display on the last module?

closed 2022-04-07 by MajicDesigns