MajicDesigns/MD_Parola #99: Compile Error ESP32 initializing Parola with SPIclass in Arduino IDE

ocraml, 2022-05-13:
## IMPORTANT
Before submitting this issue
[ x] Have you tried using the latest version of the library?
[ x] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.
I tried to set up the Hardware SPI of ESP32. The compilation failed since it could not use the 3rd constructor supplied by MD_Parola lib
that is with a defined numDevices: 
MD_Parola(MD_MAX72XX::moduleType_t mod, SPIClass &spi, uint8_t csPin, uint8_t numDevices = 1)   [C]

Reason is that the compiler considers this as a init of 
MD_Parola(MD_MAX72XX::moduleType_t mod, uint8_t dataPin, uint8_t clkPin, uint8_t csPin, uint8_t numDevices = 1)  [B]
That can also be initialized with 4 Parameters (using the default of the 5th).

Fix is simple: just exchange the order of the declaration in the header file.
(what also works is to remove the default value setting)

## Your Environment
**Library Version:** 
3.6.1
(latest code change has no difference at this point of the header)
**Arduino IDE version:** 
1.8.19

**Host OS and Version:** 
Windows 10
**CPU Hardware model/type:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Constructor using SPIclass compiles

## Actual Behaviour
Init with SPIclass not possible, compiler fails with type mismatch error

## Code Demonstrating the Issue
// Include the required Arduino libraries:
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

SPIClass * hspi = new SPIClass(HSPI);

// Hardware SPI:
#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
#define MAX_DEVICES 1
#define CS_PIN 15

// Create a new instance of the MD_MAX72XX class:
MD_Parola myDisplay = MD_Parola(HARDWARE_TYPE, *hspi, CS_PIN, MAX_DEVICES);

Insert your compilable code code here (no code snippets).
````

MajicDesigns, 2022-05-15:
Function overloading is possible by changing the types or number of arguments. The new function prototype should be ok as the parameters are different.

I think you may be confusing the compiler because it is expecting a reference to an SPIClass instance, not a pointer to it. I tested the code below and it compiles with no errors.
```
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

SPIClass hspi = SPIClass(HSPI);

// Hardware SPI:
#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
#define MAX_DEVICES 1
#define CS_PIN 15

// Create a new instance of the MD_MAX72XX class:
MD_Parola myDisplay = MD_Parola(HARDWARE_TYPE, hspi, CS_PIN, MAX_DEVICES);

void setup(void)
{
  myDisplay.begin();
}

void loop(void)
{}
```

ocraml, 2022-05-16:
I cannot reproduce the issue anymore. Both, your example and my code work fine. I must have made a mistake when I tried on Friday. Sorry for bothering with it.

closed 2022-05-16 by ocraml