MajicDesigns/MD_Parola #100: Setting for Fillers Disable?

Frank-Bemelman, 2022-08-18:
Brilliant library!

Toying around with a rather long display of 25 8x8 matrix modules and horizontal scolling of texts, I wanted to share my experience with you. It's not really an issue as such, but more a story about how I made my own fix to get what I want.

The idea is to have an endless ribbon of text. With scrollPause 0 and setScrollSpacing(3); that looks very nice. 

With texts longer than the display, the text wraps around nicely, or a newMessage rolls seamlessly into the display. Perfect.
With shorter texts, the short text is extended with fillers, and does not start again before the short text hits the leftmost columns in the display. While this is a choice of course, I prefer no fillers really. I did not find a setting to disable these fillers.

Image a series of short messages on a long display. In my case it would be something like 'Booting', 'Connecting', 'Lamp Test', 'Relay Test' and so on. If each of these messages has to scroll to the lefmost position before they are renewed, it doesn't look very appealing. Without the fillers, it may look like this:

Booting --- Booting --- Booting --- Connecting --- Lamp Test --- Lamp Test --- Relay Test --- System Up And Running --- Bucket 1 Filled --- Bucket 2 Filled -- Bucket 3 Filled ---     (and so on).

So I was very happy to find line 88 in MD_Parola_HScroll.cpp ->> _fsmState = (_countCols <= 0) ? PAUSE : PUT_FILLER;
after which I simply inserted ->> _fsmState = PAUSE;

I present all my new text with " --- " added to the end. That makes it look somewhat nicer, like the example above, to seperate the individual messages scrolling over the display in an endless fashion.

The amazing thing about this, that it took me only half an hour to make this (quick & dirty) fix to get the desired behaviour. The documentation is very very good. I have a great admiration for all your excellent work. I am not using this display on a machine, but actually use it to pimp up an old jukebox, displaying all sorts of 'data' like song being played, or radio station playing and what have you. It's all fun and you made it even more fun. Thank you!!!

Cheers,
Frank

MajicDesigns, 2022-08-19:
Thanks. Happy to hear that you are enjoying using the library.

I think the setting you were looking for is setScrtollSpacing(). Documentation entry:

<html>
<body>
<div class="memproto" style="font: bold 14px / 22px Roboto, sans-serif; border-top: 1px solid rgb(168, 184, 217); border-left: 1px solid rgb(168, 184, 217); border-right: 1px solid rgb(168, 184, 217); padding: 6px 0px; color: rgb(37, 53, 85); text-shadow: rgba(255, 255, 255, 0.9) 0px 1px 1px; background-color: rgb(223, 229, 241); box-shadow: rgba(0, 0, 0, 0.15) 5px 5px 5px; border-top-right-radius: 4px; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

void MD_Parola::setScrollSpacing (uint16_t space)  

</div><div class="memdoc" style="font: 400 14px / 22px Roboto, sans-serif; border-width: 0px 1px 1px; border-bottom-style: solid; border-bottom-color: rgb(168, 184, 217); border-left-style: solid; border-left-color: rgb(168, 184, 217); border-right-style: solid; border-right-color: rgb(168, 184, 217); padding: 6px 10px 2px; background-image: url(&quot;nav_g.png&quot;); background-repeat: repeat-x; background-color: rgb(255, 255, 255); border-bottom-left-radius: 4px; border-bottom-right-radius: 4px; box-shadow: rgba(0, 0, 0, 0.15) 5px 5px 5px; color: rgb(0, 0, 0); letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><p style="font: 400 14px / 22px Roboto, sans-serif;">Set the horizontal scrolling distance between messages for all the zones.</p><p style="font: 400 14px / 22px Roboto, sans-serif;">When scrolling horizontally, the distance between the end of one message and the start of the next can be set using this method. Default behavior is for the message to be fully off the display before the new message starts. Set to zero for default behavior.</p></div>
</body>
</html>

Frank-Bemelman, 2022-08-19:
Thank you, but I actually do use setScrollSpacing(3) not 0. 

With texts shorter than the display, and spacing set to 0 (default behaviour) the text runs off the display and when completely gone, the new message enters at the righthand side.

With texts shorter than the display, and spacing set to 1 (or 3 in my case), as soon as the text arrives at the most lefthand column, not entirely has scrolled off, the new message enters at the righthand side. Here we see a gap of empty fillers, because the display is longer than the message.

With the filler mechanism turned *off*, messages shorter than the display will be repeated without gaps. 

To demonstrate the effect, I made a video that shows/explains it better. 

I could use zones, but zones have predefined sizes and would not work here, I think..

https://www.youtube.com/watch?v=c9pWQCqJZHg

First parts is with setScrollSpacing(3) and the second part of the video is with the fillers off.

Do you see what I mean? ;-)

MajicDesigns, 2022-08-19:
The normal behaviour of Parola is to wait for the message to scroll off the screen (ie the last column of LEDs to disappear on the left) before the next message starts - this is as you observe. To do this the code counts the number of columns that are blank and compares it to the size of the display in columns. This is found in the block of code under the comment around line 124
`      // check if enough scrolled off to say that new message should start`
in the 'exit' part of the code file you modified in the original post. (BTW, now I am looking at the code I would observe that you changed the code in the 'entry' part of the animation, not the exit, so I would suggest that you change it back).

To control the number of blank columns displayed after the scrolling message, the setScrollSpacing() method sets an internal variable that overrides this around line 130
` if ((_scrollDistance != 0) && (maxCount > _scrollDistance)) maxCount = _scrollDistance;`
This feature was added some time ago for someone who was looking for the same outcome as I understand you want, so it should work.

Note that the exit part of the code is only becomes active when the message has ended the pause (which you have disabled). Where it pauses will depend on what text justification you have set for the display - it looks to me like you have set LEFT justified. In this case the message will go to the left, pause (0) and then start to exit, at which time the check for the blank columns is done.

It seems to me that you probably need to set RIGHT justification so that the text goes into exit mode when it has fully entered the display wihtout going all the way to the other end. It would probably help you debug what is going on if you also temporarily enable the pause between animation entry and exit.

Frank-Bemelman, 2022-08-19:
That's fantastic - setting the scrollAlign to PA_RIGHT is the crux of the matter here. I would never figured that out, not even by accident. PA_LEFT and setScrollSpacing(1) and that is all. My fix into the rubbish bin, all the better. You really seem to have nailed everything in this library. Apologies for wasting your time and many thanks for your kind help!

Absolutely brilliant.

closed 2022-08-20 by MajicDesigns