MajicDesigns/MD_Parola #102: How to never flip a sprite? (eg. to show a music note symbol)

x2nie, 2022-10-23:
Hi ! I am creating a [parola sprite editor](https://github.com/x2nie/parola_font_editor) (for vscode). 
it's works fine for me.

[image]

My question is: how can I use a sprite without mirroring horizontally with parolla?
explanation: Currently can I only play pacman, and this pacman heading to right when it firt running, and heading to left in second time.
I want to use an musical symbol that it should be never mirrored.

my code:
```cpp
const uint8_t F_NOTE_C = 1;
const uint8_t W_NOTE_C = 5;
const uint8_t PROGMEM note_c[F_NOTE_C * W_NOTE_C] =  // note_c
{
 0x58, 0xa4, 0xff, 0x25, 0x12,
};
```

[image]

MajicDesigns, 2022-10-23:
At the moment there is no way to do this. The sprite will be turned around by the library when the direction changes.

x2nie, 2022-10-24:
Hi, finally I found a solution without need of additional function (can work with current MD_Parola).
Well, inspired by `Parola_Animation_Catalog.ino` that show 2 different sprites for one animation, I created a flipped sprite to be shown as "not flipped" (because the second sprite will always flipped horizontally by library).

The idea is to provide 2 sprites, with the second one as flipped from first sprites
```cpp
// Sprite Definition
const uint8_t F_ROCKET = 2;
const uint8_t W_ROCKET = 11;
const uint8_t PROGMEM rocket_launch[F_ROCKET * W_ROCKET] =  // rocket
{
  0x18, 0x24, 0x42, 0x81, 0x99, 0x18, 0x99, 0x18, 0xa5, 0x5a, 0x81,
  0x18, 0x24, 0x42, 0x81, 0x18, 0x99, 0x18, 0x99, 0x24, 0x42, 0x99,
};

const uint8_t PROGMEM rocket_depart[F_ROCKET * W_ROCKET] =  // rocket flipped no-flip
{
  0x81, 0x5a, 0xa5, 0x18, 0x99, 0x18, 0x99, 0x81, 0x42, 0x24, 0x18,
  0x99, 0x42, 0x24, 0x99, 0x18, 0x99, 0x18, 0x81, 0x42, 0x24, 0x18,
};
```

The full source-code below can be shown (as live / online simulation) here : https://wokwi.com/projects/346387200800195154
[image]

full code:
```cpp
// Program to exercise the MD_Parola library
//
// Demonstrates an un-flpped sprite.
// a demo solution of question : https://github.com/MajicDesigns/MD_Parola/issues/102
//
// MD_MAX72XX library can be found at https://github.com/MajicDesigns/MD_MAX72XX
//

#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers will probably not work with your hardware and may
// need to be adapted
#define HARDWARE_TYPE MD_MAX72XX::PAROLA_HW
#define MAX_DEVICES 11

#define CLK_PIN   13
#define DATA_PIN  11
#define CS_PIN    10

// HARDWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
// SOFTWARE SPI
//MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

// Global variables
const char msg[] = "Parola Sprites";

// Sprite Definition
const uint8_t F_ROCKET = 2;
const uint8_t W_ROCKET = 11;
const uint8_t PROGMEM rocket_launch[F_ROCKET * W_ROCKET] =  // rocket
{
  0x18, 0x24, 0x42, 0x81, 0x99, 0x18, 0x99, 0x18, 0xa5, 0x5a, 0x81,
  0x18, 0x24, 0x42, 0x81, 0x18, 0x99, 0x18, 0x99, 0x24, 0x42, 0x99,
};

const uint8_t PROGMEM rocket_depart[F_ROCKET * W_ROCKET] =  // rocket flipped no-flip
{
  0x81, 0x5a, 0xa5, 0x18, 0x99, 0x18, 0x99, 0x81, 0x42, 0x24, 0x18,
  0x99, 0x42, 0x24, 0x99, 0x18, 0x99, 0x18, 0x81, 0x42, 0x24, 0x18,
};

void setup(void)
{
  P.begin();
  P.displayText(msg, PA_CENTER, P.getSpeed(), 1000, PA_SPRITE, PA_SPRITE);
  P.setSpriteData(rocket_launch, W_ROCKET, F_ROCKET, rocket_depart, W_ROCKET, F_ROCKET);
}

void loop(void)
{
  if (P.displayAnimate())
    P.displayReset();
}
```

x2nie, 2022-10-24:
or in easier to see:
[image]

closed 2022-10-24 by x2nie