MajicDesigns/MD_Parola #105: Display Sprite data as a simple character without scroll - not issue just a question about library working

Sentinel8000, 2022-12-25:
You make a great job with these librarys.

I try make a weather station and use animated icons example:

const uint8_t F_clearDay = 3;
const uint8_t W_clearDay = 8;

const uint8_t PROGMEM clearDay[F_clearDay * W_clearDay] =
{ B00000000, B00000000, B00011000, B00111100, B00111100, B00011000, B00000000, B00000000,
  B00000000, B01000010, B00011000, B00111100, B00111100, B00011000, B01000010, B00000000,
  B10010001, B01000010, B00011000, B00111100, B00111100, B00011000, B01000010, B10001001,
};

Currently, i can use this with this command:

 P.displayText(" ", PA_LEFT, 100, PAUSE_TIME, PA_NO_EFFECT, PA_SPRITE);                       
 P.setSpriteData(clearDay, W_clearDay, F_clearDay, clearDay, W_clearDay, F_clearDay); 

So i need first write out a blank character and set a  PA_SPRITE parameter. Pre defined sprite (animated sun) start scrolling from left to right.

But what if, if i want display this sprite just on the 1. panel from the 4, without any scrolling.

There is any way display the sprite without any scroll? or the library can only display sprite just with scrolling?

thank you and merry X-mas

MajicDesigns, 2022-12-25:
A sprite by definition is the display of a number of bitmaps one after the other to create an animation. If you do not want any animation then the display of one frame (bitmap) of the sprite is just like the display for one character. There are functions to do this in the library, see the CustomChars example.

Also, if all you are displaying is some characters with no animations required at all, then the MD_MAX72xx library by itself would be more memory efficient as there will not be the overhead of the animation functions.

Sentinel8000, 2022-12-25:
You have right the easiest way simple define a customs font and display.

I try use not the simpliest way.

The idea comes from a completted github project where the creator used the ledcontrol library:

[https://github.com/jconenna/WiFiWeatherDisplay/blob/master/WiFiWeatherDisplay.ino](url)

if i good see in the code was the setrow command used, ,,play" the sprite without any scroll:

void displayImage(byte image[])
{
  for (unsigned int i = 0; i < 8; i++)
    lc.setRow(0, i, image[i]);
}

The question is, with parola library somehow possible reptoduce this, example playing with the zone handling?

Also im sure impossible use both library in same time.

Here you can see the code in action:

[https://www.youtube.com/watch?v=qIY1XLZ9a7k&t=43s](url)

Sentinel8000, 2022-12-25:
I have thinked also a alternate solution - this can be reproduced simple way, creating more than one font and changing these and repeat, however this can be long work creating more fonts to reproduce ,,animation"  effect:)

MajicDesigns, 2022-12-25:
You can substitute just one character in the current font using a run time call.

You can also use the setRow() method in MD_MAX72xx.

Sentinel8000, 2022-12-26:
With the substitute i can add just one character, to simulate the sprite i need define more characters if im good understand the working.

The second solution im sure will work, but i can use in this case just the MD_MAX72xx functions/commands, i need forget MD_Parola features. 

Here i have a another example, but maybe this is also just more defined fonts  in a circle, code was not shared:

[[MD_MAX72xx](https://youtu.be/5bRLm78zYHE?t=38)](url)

thanks for the fast answers and also merry xmas for you

Sentinel8000, 2022-12-30:
Because you have answered fast and you have wroted this library, i think important to share my solution what is i think not a practical solution, but works - so i dont need ,,convert" the pre-defined animations to fonts.

#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
#define HARDWARE_TYPE_P MD_MAX72XX::FC16_HW

```
#define CS_PIN    5  // or SS
#define SS_PIN    15  // or SS

MD_Parola P = MD_Parola(HARDWARE_TYPE_P, CS_PIN, 4);
MD_Parola Q = MD_Parola(HARDWARE_TYPE_P, SS_PIN, 4);

MD_MAX72XX mx = MD_MAX72XX(HARDWARE_TYPE, SS_PIN, 4);
.
.
.
void displayImage(byte image[])
{
  
  for (unsigned int i = 0; i < 8; i++)

    mx.setRow(3, i, image[i]); //play pre-defined animations on 4. block
    
}
```

With  these definitions, i can control just 1 ,,block" and also i have not lost the practical scrolling feature of parola.

I need choose for MD_MAX72XX different HW type because of the nature of the defined animation was 90 degree rotatted. 

I have connected the 2x 4 matrix using esp32 two SPI, so was for myself easier control both matrix against using zones.

Maybe not good idea, but lookslike there is no problem using parallel MD_max72xx and MD_parola and overlap the handled zones. 

I have found after the 90 degree problem, another one, because these sprite animations running continous, to not stutter the scrolling i have used esp32 dual core feuture. Core 1 running the MD_MAX72xx animations with setrow the core 2 running the parola scrollng.

Here is the result:

[https://youtu.be/JvDd3fo5G94](url)

thanks for you work

MajicDesigns, 2022-12-31:
OK if this works for you. However please note that:

1. It is not safe to use a separate MD_MAX72xx object. You already have one being used by the MD_Parola library and you can access this copy. You should use
```
MD_MAX72XX *mx;

mx = getGraphicObject();   // in setup or anywhere before you use this

mx->setRow(...); /// when you want to use the MD_MAX72xx methods
```

2. You can change the hardware type at run time using the MD_MAX72xx library (see MD_MAX72xx_Dynamic_HW example) so you do not need to declare more than one MD_Parola object.

closed 2022-12-31 by MajicDesigns