MajicDesigns/MD_Parola #109: strrev missing??

Guenni75, 2023-01-29:
Hello.

I installed MD_Parola and MD_MAX72xx today.

I wanted to try Parola_scrolling_vertical but i get the error:

Parola_Scrolling_Vertical.ino:160:7: error: 'strrev' was not declared in this scope; did you mean 'strsep'?

Other examples are working fine.

MajicDesigns, 2023-01-29:
The strrev() function (used to reverse the given string) is a built-in function in C and is defined in string.h header file, which is a standard header file for all C++ compilers.

Please check that you have string.h on your computer.

What development environment are you using?

Guenni75, 2023-01-30:
I work with the Arduino IDE 2.0.3.

MajicDesigns, 2023-01-30:
IDE 2.0.3 under Windows 10 with target Arduino Uno compiles cleanly on my system.

Guenni75, 2023-01-31:
I compile with a WEMOS D1 (esp8266).
The problem is, strrev is no standard C library function.

[image]

Guenni75, 2023-01-31:
I tried to reverse the Array by myself.
I changed BUF_SIZE to 7.

[image]

The Monitor shows everything is ok, BUT
I don't understand, why is reverseMessage empty??????
And newMessage is an Array! Why can i print it so simple with Serial.print ???

MajicDesigns, 2023-01-31:
Your reverse function is logically flawed. The string is generally not the size of the buffer (so you should use strlen() not BUF_SIZE) and you are not handling the terminating nul character properly in the reversal process. The standard strrev() does the reverse in the same buffer (ie, no additional memory required).

Here is the code for the strrev() function from the libraries:
```
void strrev(unsigned char *str)
{
  int i;
  int j;
  unsigned char a;
  unsigned int len = strlen((const char *)str);
   
  for (i = 0, j = len - 1; i < j; i++, j--)
  {
  a = str[i];
  str[i] = str[j];
  str[j] = a;
  }
}
```

Also, for future reference, please include text and not graphics when including code or debug output in your messages as these are easier to handle if we need to cut/paste for testing.

Guenni75, 2023-02-01:
Thanks.
I will try this.

closed 2023-02-04 by MajicDesigns