MajicDesigns/MD_Parola #112: Interrupt text effect in anytime

uronito, 2023-03-22:
Hi, 

First of all I wanted to thank you for your hard work for this great library.

Would it be possible to stop a vertical text scroll effect at any time?

It would be like interrupting the animation at any time

Sometimes I mount a text effect with another.

MajicDesigns, 2023-03-22:
If you stop calling .displayAnimate() the display will stop animating and 'freeze in place. If you want to clear the current animation you can clear the display/zone using .displayClear() and reset the animation sequence using .displayReset(). Look at the "Methods for core object control" section of the library help file.

closed 2023-03-22 by MajicDesigns

uronito, 2023-03-23:
don't work, here is my code

``
const char* OTA_hostname    = "Panel-led";
const char* OTA_password    = "";      // Don't forget to set your settings!
#include <ArduinoJson.h>

#include <MD_Parola.h>
#include <Time.h>
#include <MD_MAX72xx.h>            // For LED matrix display
#include <SPI.h>                   // For connecting matrix display
#include <PubSubClient.h>          // For MQTT
#include <ESP8266WiFi.h>           // For WiFi connection with ESP8266
#include <WiFiUdp.h>               // For NTP Signal fetch
#include <EasyNTPClient.h>         // For NTP Signal read https://github.com/aharshac/EasyNTPClient
#include <TimeLib.h>// For converting NTP time https://github.com/PaulStoffregen/Time.git
#include <ArduinoOTA.h>
#include "OTA_update.hpp"
#include "Font_Data.h"
#include "Settings.h"
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 8
#define MAX_ZONES 3

unsigned long timing;

#define CLK_PIN   14
#define DATA_PIN  13
#define CS_PIN    15
char  szMesg[21] = "";
int pirSensor = 5;                               // Pin, where PIR sensor is connected
bool nuevo_mensaje = false;
#define SPEED_TIME  45
#define PAUSE_TIME  1000
bool encendido;
bool tres_zonas;

WiFiUDP udp;
EasyNTPClient ntpClient(udp, NTP_SERVER, TZ_SEC);

MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

MD_MAX72XX *mx;

void(* resetFunc) (void) = 0;       // declare reset function @ address 0

WiFiClient espClient;               // needed for MQTT
PubSubClient mqtt_client(espClient);     // needed for MQTT

// Global variables
uint8_t  curText;

uint8_t degC[] = { 6, 3, 3, 56, 68, 68, 68 };

String tempvar;

char actualtime[16];
char actualseconds[16];
char actualday[90];
char outdoortemp[2048];
char pooltemp[512];
char indoortemp[512];
char zambrettiswords[512];
char trend[512];
char *mqtterror = "MQTT Error";
char *wifierror = "WiFi Error";
char mensaje_mosquito[2048];
int verde = D6;
int rojo = D4;

[... 617 lines cut ...]
           
         

        }
      }
    }

  }
         
}

boolean summertime_EU(int year, byte month, byte day, byte hour, byte tzHours)
// European Daylight Savings Time calculation by "jurs" for German Arduino Forum
// input parameters: "normal time" for year, month, day, hour and tzHours (0=UTC, 1=MEZ)
// return value: returns true during Daylight Saving Time, false otherwise
{
  if (month < 3 || month > 10) return false; // keine Sommerzeit in Jan, Feb, Nov, Dez
  if (month > 3 && month < 10) return true;  // Sommerzeit in Apr, Mai, Jun, Jul, Aug, Sep
  if (month == 3 && (hour + 24 * day) >= (1 + tzHours + 24 * (31 - (5 * year / 4 + 4) % 7)) || month == 10 && (hour + 24 * day) < (1 + tzHours + 24 * (31 - (5 * year / 4 + 1) % 7)))
    return true;
  else
    return false;
}

``

MajicDesigns, 2023-03-23:
Sorry, but I really don't have time to analyse your code. If you can reduce it to a MINIMAL code set that just uses the Parola library and still does not work then I can replicate this and check what is happening.