MajicDesigns/MD_Parola #113: Use of MD_parola

R0lleston, 2023-06-28:
Hi Marco

Apologies for post under issues.
I’m trying to ascertain suitability of your library for a stupidly simple project (for a programmer anyway :) )

lighting up 22 arcade buttons that contain non addressable standard 5mm leds.

I’m no programmer so I need something straight forward that will allow me to create a pattern that loops continuously until a push button moves it on to another pattern.

I plan on using a nano, with the max7219 to nterface with the 22 leds.

MajicDesigns, 2023-06-28:
You can use the MD_MAX72xx library to address up to 64 leds individually - essentially one module.  You can also use a boil extender (specialized hardware or the 595 IC) to do the same and it would probably be easier programming.

R0lleston, 2023-06-28:
Cheers, thanks for the steer in the right direction

closed 2023-06-28 by MajicDesigns