MajicDesigns/MD_Parola #114: [QUESTION] turn specific pixel on or off?

othmar52, 2023-07-25:
Hi,
Currently i try to [build a speedometer](https://github.com/othmar52/table-soccer-speedometer) for a table soccer that has 8 light barriers per goal  

i use `displayZoneText()` with `PA_NO_EFFECT`  
For calibration purposes i'd like to switch a specific pixel for each light barrier on or off to reflect its state.

is this posible with the MD_Parola library without using custom characters?  
the code for those custom characters would be a huge mess because of permutation of all possible states of the 16 light barriers

### pseudo code
```c++
// render the text in all zones
// as we dont use an effect it renders immediatly
Display.displayAnimate();

// render the state of each light barrier
if (lightBarrier8 == HIGH) {
  Display.turnPixelOff(x8, y8);
} else {
  Display.turnPixelOn(x8, y8);
}
```

thank you

MajicDesigns, 2023-07-25:
Just use the MD_MAX72xx library setPoint() directly to turn on the pixels.

If you are doing this as a once off testing code, just use MD_MAX72xx. If you are using it in your code always then you get the MD_MAX72xx library pointer used by MD_Parola using the getGraphicObject() method.

closed 2023-07-25 by MajicDesigns

othmar52, 2023-07-29:
thank you for your fast help!  
in case there are some other arduino noobs like me out there
```c++
MD_Parola Display = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
MD_MAX72XX *pM;

  // use setPoint() to turn on/off a specific pixel
  pM = Display.getGraphicObject();
  pM->setPoint(5, 0, true);
```