MajicDesigns/MD_Parola #115: Remanant residual text on Double_Height on 2x19 matrix

angeoliver, 2023-07-31:
Hello,

Thanks for your impressive work.

I have an unsolved problem with the **Double_Height** feature, on a **2x19 matrix** (38 modules).
The result is good but there is a final frozen/remanant residual text on only zone 1, on the 6 last left modules.
Maybe a buffer overflow with this large matrix ?
All examples "Double_Height" scripts have this problem on my configuration, for example Parola_Double_Height_v2.ino

**You can see my video of this issue : https://youtu.be/eFnl-EL3cHk**

I can't resolve this issue by myself after many hours of research on the 2 libraries. 
On standard features (not Double_Height) this problem is not present (ex: Parola_Sprites_Library with one zone).
Thank you very much for your advice or help please.

Regards,

Olivier

Libraries Versions: MD_Parola 3.7.1 / MD_MAX72XX 3.4.1

Arduino IDE version: 1.8.13
Host OS and Version: Linux Mint
CPU Hardware model/type: i3-4170

Script of the test: Parola_Double_Height_v2.ino
My tuning:
18: #define HARDWARE_TYPE MD_MAX72XX::ICSTATION_HW
19: #define MAX_DEVICES 19
38: #define SPEED_TIME  10

MajicDesigns, 2023-08-01:
Looks like you are running an Arduino Uno?
Also it looks like the text stops at a module boundary (6 modules if I can count correctly but it is difficult from the video). If this is the case then I would be looking at an electrical/connection problem.

- How are the matrices wired? Are the matrices in the top left of the display in the middle of the display modules or are they at the end? 

- How are you distributing the digital signals (except for DAT which is module to module)?

angeoliver, 2023-08-01:
Thanks for your reply.
I use a Arduino Nano (on back of the matrix).
New elements: with default scroll right left (#define SCROLL_LEFT 0) the problem is present but it's 100% OK with scroll right (#define SCROLL_LEFT 0).
I confirm: 6 modules 8x8 are concerned by the problem: the 6 last of the chain.
I use PCB of 4 modules 8x8 therefore no connection problem I think.
A remark: the problem concern the 6 last modules on a total of 38 modules: (38-6)*8 = 256 columns. Pb if column > 256. It's a coincidence ? Or overflow of a uint8_t variable ?
Thanks,
Olivier

MajicDesigns, 2023-08-01:
"I use PCB of 4 modules 8x8 therefore no connection problem I think."
Are you able to rearrange the modules in a different order. Does the problem follow the modules or does it stay in the same place (ie, last 6)?

"A remark: the problem concern the 6 last modules on a total of 38 modules: (38-6)*8 = 256 columns. Pb if column > 256. It's a coincidence? Or overflow of a uint8_t variable?"
Maybe, but scroll left/right code is the same code with different branching. 

If you can confirm first point it will tell us whether the fault is software related.

Also can you confirm if the modules are joined as
4321  < top line order (incoming connection at 1)
4321  < bottom line order (incoming connection at 1, outgoing connection at 4 to 1 above)
or
1234
4321
an also whether you are distributing the signals or just relying on daisy chain from first to last? (see [this blog post](https://arduinoplusplus.wordpress.com/2018/07/13/fun-and-games-and-a-new-library/))

angeoliver, 2023-08-05:
Hello,

I found a correction for my issue :

Libray **MD_MAX72xx.h** v**3.4.1**, line **661** replace :
uint8_t getColumn(**uint8_t** c) { return getColumn((c / COL_SIZE), c % COL_SIZE); };
by
uint8_t getColumn(**uint16_t** c) { return getColumn((c / COL_SIZE), c % COL_SIZE); };

The final frozen/remanant residual text on columns > 255 is corrected.

Thanks for you time et your work,

Olivier

MajicDesigns, 2023-08-05:
Release v3.4.2

closed 2023-08-05 by MajicDesigns