MajicDesigns/MD_Parola #120: Display Clear is not working

DeveloperLg1, 2023-12-08:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
I try to set displayDelar() once trigger analogRead(A0) in esp8266 wemos lolin D1 R2

## Your Environment
Library Version:3.7.1
Arduino IDE version:1.8.19
Host OS and Version:?
CPU Hardware model/type: Lolin wemos D1 R2 ESSP8266

## Steps to Reproduce
Display works without problems at first. one moment i try to set displayClear() isn't not working. could you help me to solve?

##This Video
https://drive.google.com/file/d/1qDMwqela4fgi9czhUpzcMsR_SLy7rNhE/view?usp=drivesdk 

## Code Demonstrating the Issue
````
///*
//  ReadAnalogVoltage
//
//  Reads an analog input on pin 0, converts it to voltage, and prints the result to the Serial Monitor.
//  Graphical representation is available using Serial Plotter (Tools > Serial Plotter menu).
//  Attach the center pin of a potentiometer to pin A0, and the outside pins to +5V and ground.
//
//  This example code is in the public domain.
//
//  https://www.arduino.cc/en/Tutorial/BuiltInExamples/ReadAnalogVoltage
//*/
//
//// the setup routine runs once when you press reset:
//void setup() {
//  // initialize serial communication at 9600 bits per second:
//  Serial.begin(115200);
//}
//
//// the loop routine runs over and over again forever:
//void loop() {
//  // read the input on analog pin 0:
//  int sensorValue = analogRead(A0);
//  // Convert the analog reading (which goes from 0 - 1023) to a voltage (0 - 5V):
//  float voltage = sensorValue * (5.0 / 1023.0);
//  // print out the value you read:
//  if(voltage==5.0){
//    Serial.println(voltage);
//  }
//  Serial.println(sensorValue);
//}

//LED Matrix
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW         // see video number one about this you have a choice of 4 hardware types to select
#define MAX_DEVICES 4
#define DATA_PIN D5
#define CS_PIN D6
#define CLK_PIN D7

MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);   // see video 1 why we have called this "P"

void loop()
{
  if ((WiFi.status() == WL_CONNECTED))
  {
    int sensorValue = analogRead(A0);
    // Convert the analog reading (which goes from 0 - 1023) to a voltage (0 - 5V):
    float voltage = sensorValue * (5.0 / 1023.0);
    // print out the value you read:
    //Serial.print("voltage"); Serial.println(voltage);
    if (voltage > 3.0) {
      StopLamp = "STOP";
      if (P.displayAnimate())
        P.displayText(StopLamp.c_str(), animList[i].align, animList[i].speed,
                      animList[i].pause, animList[i].effecIn, animList[i].effecOut);
      //Serial.println(voltage);
    }
    else {
      RunAnimations();
    } 
    server.handleClient();
  }
}

````

MajicDesigns, 2023-12-08:
You should be providing a separate 5v power supply for the matrices. They often don’t work properly using 3v from the cpu board.

DeveloperLg1, 2023-12-18:
Thanks for reply buddy.
the issue has been solved.

closed 2023-12-18 by DeveloperLg1