MajicDesigns/MD_Parola #121: issue on P.displayAnimate() parola

DeveloperLg1, 2023-12-18:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
I try to playing animate

## Your Environment
**Library Version:3.7.1** 
**Arduino IDE version:1.8.19** 
**Host OS and Version:OS win 10** 
**CPU Hardware model/type:Lolin wemos D1 R2 ESSP8266 / Node MCU V3** 

## Steps to Reproduce
I'm try to run animate character with font arabic and roman. and have issue on Exception Decoder. then try investigate the issue to EPS exception decoder. path issue is 
0x4020c4e5: MD_PZone::commonPrint() at <home>libraries\MD_Parola\src\MD_Parola_Print.cpp line 47

result output terminal.
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffcc0 end: 3fffffc0 offset: 01a0
3ffffe60:  3ffeff30 3fff1049 3fff0ff0 4020c4e5  
3ffffe70:  0000000b 3ffeff38 401002bd 3ffeff38  
3ffffe80:  3ffeff30 00000001 3fff0ff0 3ffeff38  
3ffffe90:  3ffeff30 00000001 3fff0ff0 4020bbe9  
3ffffea0:  3ffeff30 00000032 3fff0ff0 4020aee5  
3ffffeb0:  3ffeff30 00000000 00000000 4020b2b8  
3ffffec0:  40212001 000003e8 3ffef8f8 3fff010c  
3ffffed0:  3ffeff30 3fff0154 3ffef777 402029ce  
3ffffee0:  007a1200 c1f52b76 4bc6a700 00000000  
3ffffef0:  40105c75 3fffff10 00000001 40213f0c  
3fffff00:  40249276 3fff0154 0000005f 00000003  
3fffff10:  4024a463 00000000 00001388 3fff010c  
3fffff20:  3fff00d3 3ffef776 3ffef777 40202eea  
3fffff30:  3fff00d3 0000005f 3fff0154 4020921b  
3fffff40:  00000002 feefeffe feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe 3fff1424  
3fffff60:  003f003f 8a003231 3fff13ec 0020002f  
3fffff70:  8a000000 feefeffe feefeffe feefeffe  
3fffff80:  0000005f 00000002 00000000 feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff02d0  
3fffffa0:  3fffdad0 00000000 3fff02bc 40211a0c  
3fffffb0:  feefeffe feefeffe 3ffe8680 40100cc5  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Result from esp exception decoder
[image]

 

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
  static uint8_t i = 5;
  Serial.println("x3 : ");
  if (P.displayAnimate())                                                 // animates and returns true when an animation is completed
  {
    Serial.println("x4 : ");
    if (xx == i)
    {
      i = 1; Serial.println("x5 : ");
    }
    // prevTimeAnim = millis();
    if (i == ARRAY_SIZE(animList))
      i = 1;                                  // reset loop index

    Serial.println("x6 : ");
    Serial.print("pText : "); Serial.println(animList[i].pText);
    Serial.print("textalignment : ");   Serial.println(animList[i].align);
    Serial.print("speed : ");   Serial.println(animList[i].speed);
    Serial.print("pause : ");   Serial.println(animList[i].pause);
    Serial.print("effecIn : ");    Serial.println(String(animList[i].effecIn));
    Serial.print("effecOut : ");    Serial.println(String(animList[i].effecOut));
    Serial.print ("character : "); Serial.println(animList[i].character);

    Serial.println(i);
    Serial.print ("Total loop animate"); Serial.println(xx);

    if (animList[i].character != "0") {

      if (animList[i].character == "Basmallah")
      {
        P.setFont(M[1].pFont);
        P.displayText(M[1].pMsg, alignList[i].textalignment, animList[i].speed,  animList[i].pause, animList[i].effecIn, animList[i].effecOut);
        P.setTextBuffer(M[1].pMsg);
        P.setTextEffect(animList[i].effecIn, animList[i].effecOut);
        P.displayReset();
      }
      if (animList[i].character == "Lailah")
      {
        P.setFont(M[1].pFont);
        P.displayText(M[2].pMsg, alignList[i].textalignment, animList[i].speed,  animList[i].pause, animList[i].effecIn, animList[i].effecOut);
        P.setTextBuffer(M[2].pMsg);
        P.setTextEffect(animList[i].effecIn, animList[i].effecOut);
        P.displayReset();
      }

      P.displayReset();
      i++;

    } else {

      P.setFont(M[0].pFont);
      P.displayText(animList[i].pText.c_str(), alignList[i].textalignment,  animList[i].speed,
                    animList[i].pause, animList[i].effecIn, animList[i].effecOut);
      P.setTextBuffer(animList[i].pText.c_str());
      P.setTextEffect(animList[i].effecIn, animList[i].effecOut);

      P.displayReset();
      i++;
      delay(1000);
      Serial.println("x1 : ");
    }
  }

  Serial.println("x2 : ");
````

MajicDesigns, 2023-12-18:
The text buffer must persist while the animation is being completed. In the case where you are using the c_Str conversion for a String this is unlikely to be the case.

DeveloperLg1, 2023-12-18:
> The text buffer must persist while the animation is being completed. In the case where you are using the c_Str conversion for a String this is unlikely to be the case.

Can we adjust the settbuffer to null or ""?

MajicDesigns, 2023-12-19:
You should adjust to a valid c-string that remains in scope while the animation is active.

closed 2023-12-24 by MajicDesigns