MajicDesigns/MD_Parola #123: access the dot display buffer before animation ?

woeperbe, 2024-01-01:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Host OS and Version:** 
**CPU Hardware model/type:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your compilable code code here (no code snippets).
````

MajicDesigns, 2024-01-01:
Please read the response to issue #114.

woeperbe, 2024-01-01:
That doesn't answer my question.
I need to be able to animate the display with my own bit pattern as in the picture ...

MajicDesigns, 2024-01-02:
It does answer the question "i would like to insert in the display buffer just before the animation my own bit pattern" - you use getGraphicObject() and use the MD_MAX72xx function to draw the graphic you want. If you now want to animate just this 'graphic' you will do it your code, as Parol will have no knowledge of this extra stuff on the display.

Without knowing the details of what you are trying to do I cannot suggest much more specific solution. Maybe describe what you are trying to do rather than the solution you think you need.

woeperbe, 2024-01-02:
I will try this today
Thanks for your beautifull library and the support at wrapspeed
Happy new-year
Greetings,
Marc Buffet

closed 2024-01-19 by MajicDesigns