MajicDesigns/MD_Parola #124: Software SPI not working on Arduino Nano ESP32

Philoneous, 2024-01-05:
## IMPORTANT
Before submitting this issue
[x ] Have you tried using the latest version of the library?
[x ] Have you checked this has not already been submitted and/or resolved?
[x ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Software SPI does not work on Arduino Nano ESP32

## Your Environment
**Library Version:3.7.1** 
**Arduino IDE version:2.2.1** 
**Host OS and Version:Win 10** 
**CPU Hardware model/type:** 

## Steps to Reproduce
Whenever I use the constructor for software SPI (MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES)) the Matrix doesn't work. Using the Hardware constructor and the according pins works, however.

## Expected Behaviour
Software SPI should be functional when defining the pins and using the according constructor.

## Actual Behaviour
matrix remains blank

## Code Demonstrating the Issue
````
// Program to demonstrate the MD_Parola library
//
// Simplest program that does something useful - Hello!
// Uses the Arduino Print Class extension
//
// MD_MAX72XX library can be found at https://github.com/MajicDesigns/MD_MAX72XX
//

#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers will probably not work with your hardware and may
// need to be adapted
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 4

#define CLK_PIN   4
#define DATA_PIN  5
#define CS_PIN    6

// Hardware SPI connection
//MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
// Arbitrary output pins
MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

void setup(void)
{
  P.begin();
  P.print("Hello!");
}

void loop(void)
{
}

MajicDesigns, 2024-01-27:
Following investigation was unable to replicate this problem.

Tested with the code below and hardware (Arduino SPI library) and software (bitbang) SPI worked ok on both AVR and ESP32 platforms.

````
#include <MD_Parola.h>
#include <MD_MAX72xx.h>

#define ESP32_TEST

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 4

#ifdef ESP32_TEST
// ESP32 Definitions
#define CLK_PIN   18
#define DATA_PIN  23
#define CS_PIN    5
#else
// AVR Definitions
#define CLK_PIN   13
#define DATA_PIN  11
#define CS_PIN    10
#endif

// Hardware SPI connection
//MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);
// Arbitrary output pins
MD_Parola P = MD_Parola(HARDWARE_TYPE, DATA_PIN, CLK_PIN, CS_PIN, MAX_DEVICES);

void setup(void)
{

  Serial.begin(115200);
  Serial.println("-----");
  Serial.print("MOSI: ");   Serial.println(MOSI);
  Serial.print("MISO: ");   Serial.println(MISO);
  Serial.print("SCK: ");    Serial.println(SCK);
  Serial.print("SS: ");     Serial.println(SS);

  if (!P.begin()) Serial.println("!!P.begin() error");
  P.print("Hello!");
}

void loop(void)
{
}
````

MajicDesigns, 2024-02-04:
Close as no response from OP

closed 2024-02-04 by MajicDesigns