MajicDesigns/MD_Parola #127: Show RtcDS3231.h time on Max7219

8266er, 2024-03-28:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

Hello Marco.
Thanks for your reply.

I want to display time, date and year (RtcDS3231.h) with scroll effect (MD_Parola.h, MAX72xx.h) on a 32×8 MAX7219 matrix with a NodeMCU8266.

Everything is displayed correctly in the serial monitor of the Arduino IDE.

But time, date and year are not displayed on the Max7219 matrix ;-(

What am I doing wrong (line 128 to 144)????

I would be very happy if you could take a look at the skecht and help me with these lines.

Here the code:
````
Insert your compilable code code here (no code snippets).
````
Describe your issue here.

## Your Environment
**I am using MD_Parola V 3.7.2, MD_MAX72xx V 3.5.2.** 
**Arduino IDE version:1.8.13** 
**Host OS and Version: Ubuntu Linux 22.04** 
**CPU Hardware model/type: 4x Intel cor i3-5005U 2.00GHz** 

## Code Demonstrating the Issue
````

#include <DFRobotDFPlayerMini.h>
#include <Arduino.h>
#include <Wire.h>          // normal Hardware Wire
#include <SoftwareWire.h>  // for software wire
#include <RtcDS3231.h>     // RTC library by Makuna
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include <SoftwareSerial.h

#define HARDWARE_TYPE MD_MAX72XX::ICSTATION_HW
//#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
//#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
#define MAX_DEVICES 4

// for NODE MCU
#define CLK_PIN  14 // D5  // 14
#define DATA_PIN  13 // D7  // 13
#define CS_PIN    15 // D8  // 15

//#define CLK_PIN   14
//#define DATA_PIN  13
//#define CS_PIN    15

// CONNECTIONS DS3231:
// DS3231 SDA --> SDA
// DS3231 SCL --> SCL
// DS3231 VCC --> 3.3v or 5v
// DS3231 GND --> GND  

//RtcDS3231 rtcObject;                 // Bei LibVers kleiner 2.0.0 Hardware Wire
RtcDS3231<TwoWire> rtcObject(Wire);  // Bei LibVers 2.0.0 und größer Hardware Wire

//SoftwareWire myWire(SDA, SCL);       // for software wire
//RtcDS3231<SoftwareWire> Rtc(myWire); // for software wire

[... 194 lines cut ...]
   else if (( currentTime.Hour == 3 ) && ( currentTime.Minute == 0 ) && ( currentTime.Second == 0 ))  {  // 03 Uhr
//   if (( currentTime.Hour() == 3 ) && ( currentTime.Minute() == 0 ) && ( currentTime.Second() == 0 ))    // 03 Uhr
    myDFPlayer.play(0004);
  }
   else if (( currentTime.Hour == 2 ) && ( currentTime.Minute == 0 ) && ( currentTime.Second == 0 ))  {  // 02 Uhr
//   if (( currentTime.Hour() == 2 ) && ( currentTime.Minute() == 0 ) && ( currentTime.Second() == 0 ))    // 03 Uhr
    myDFPlayer.play(0003);
  }
   else if (( currentTime.Hour == 1 ) && ( currentTime.Minute == 0 ) && ( currentTime.Second == 0 ))  {  // 02 Uhr
//   if (( currentTime.Hour() == 1 ) && ( currentTime.Minute() == 0 ) && ( currentTime.Second() == 0 ))    // 02 Uhr
    myDFPlayer.play(0002);
  }
   else if (( currentTime.Hour == 0 ) && ( currentTime.Minute == 0 ) && ( currentTime.Second == 0 ))  {  // 00 Uhr
//   if (( currentTime.Hour() == 0 ) && ( currentTime.Minute() == 0 ) && ( currentTime.Second() == 0 ))    // 00 Uhr
    myDFPlayer.play(0001);
  }
}

````

Best regards

Paul

[Max7219_RTC_MP3Clock.zip](https://github.com/MajicDesigns/MD_Parola/files/14786132/Max7219_RTC_MP3Clock.zip)

MajicDesigns, 2024-03-29:
The code
`
  P.print(( currentTime.Hour) && ( currentTime.Minute));
`
Is not corrrect. In C++ && operator is a boolean operator. You are using this operator correctly in your if statements at the end of the loop() function.

MD_Parola requires you to pass through a c-string for it to display and the c-string must remain in scope for the duration of the animation. You are creating a c-string in str above these lines to display debug on the serial monito - you should initially try and display this c-string to the matrix to confirm that your code is correct.

In order to display with scrolling you cannot use .print(). You will need to restructure the code to allow the animation to take place (no delay() and a free running loop() function). The Parola_Zone_TimeMsg example is close to what you need to do - just ignore the zone related stuff and focus on how the loop() code is structured to display different messages in sequence.

MajicDesigns, 2024-03-29:
You have not eliminated the code that references zones completely. The below should display the time, updated each second. If you are still getting 00:00 then you may have comms issues to the clock module.
`
void loop(void)
{
...
  P.displayAnimate();

  // Finally, adjust the time string if we have to
  if (millis() - lastTime >= 1000)
  {
    lastTime = millis();
    getTime(szTime, flasher);
    flasher = !flasher;

    P.displayReset();
  }
 ...
}
`

8266er, 2024-03-30:
Hello Marco.

Thank you very much for your renewed help!!!

Now the time is displayed correctly, but when a full hour is reached
and the MP3 file is called up, the clock stops at the hour.

I have no idea how to solve this.

I would be very happy if you could help me ONE LAST TIME ;-)

Best regards

Paul

````

//Vorlage MD_Parola_Zone_TimeMsg
// Use the DS1307 clock module
#define USE_DS1307 1  // 0

// Header file includes
#include <DFRobotDFPlayerMini.h>
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <MD_DS1307.h>
#include <Wire.h>
#include <SPI.h>

#include "Font_Data.h"

// Hardware deffinition MAX7219
#define HARDWARE_TYPE MD_MAX72XX::ICSTATION_HW
//#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
//#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
//#define HARDWARE_TYPE MD_MAX72XX::PAROLA_HW

#define MAX_DEVICES 4

// for NODE MCU
#define CLK_PIN   14  // D5
#define DATA_PIN  13  // D7
#define CS_PIN    15  // D8

// CONNECTIONS DS1307:
// DS3231 SDA --> SDA
// DS3231 SCL --> SCL
// DS3231 VCC --> 3.3v or 5v
// DS3231 GND --> GND

#define myRTC RTC ///< Library created instance of the RTC class

// DF Player Mini
//SoftwareSerial mySerial1(1, 3);
SoftwareSerial mySerial1(3, 1);
//NodeMCU TX (GPIO1) an DF Player mini RX, NodeMCU RX (GPIO3) an DF Player mini TX
DFRobotDFPlayerMini myDFPlayer;  // DF Player definieren
/*
  Verzeichnis auf SD Karte muss „MP3“ heissen!
  Dieses Verzeichnis muss exakt als „MP3“ benannt werden.
  HBis zu 3000 Dateien mit dem Namensschema 0001.mp3 (bzw. wav), 0002.mp3, ….. 3000.mp3.
  Alle Dateien sind direkt ansprechbar und keine Probleme mit der Reihenfolge.
*/

void myDFPlayerSetup() {
  myDFPlayer.setTimeOut(500);
  myDFPlayer.volume(18);  //Lauttstärke Auswahl 0 bis 30
  myDFPlayer.EQ(DFPLAYER_EQ_BASS);
[... 179 lines cut ...]
  }
  else if (( RTC.h == 5 ) && ( RTC.m == 0 ))  {  // 05 Uhr
    myDFPlayer.play(6);
  }
  else if (( RTC.h == 4 ) && ( RTC.m == 0 ))  {  // 04 Uhr
    myDFPlayer.play(5);
  }
  else if (( RTC.h == 3 ) && ( RTC.m == 0 ))  {  // 03 Uhr
    myDFPlayer.play(4);
  }
  else if (( RTC.h == 2 ) && ( RTC.m == 0 ))  {  // 02 Uhr
    myDFPlayer.play(3);
  }
  else if (( RTC.h == 1 ) && ( RTC.m == 0 ))  {  // 02 Uhr
    myDFPlayer.play(2);
  }
  else if (( RTC.h == 0 ) && ( RTC.m == 0 ))  {  // 00 Uhr
    myDFPlayer.play(1);
  }

  //+++   */
}

````
[Max7219_RTC_MP3Clock.zip](https://github.com/MajicDesigns/MD_Parola/files/14812651/Max7219_RTC_MP3Clock.zip)

MajicDesigns, 2024-03-30:
Does the clock restart when the tune stops playing? Does th tune play at all?

It seems to me that your logic should be that the minutes and the seconds should be == 0 to play the tune otherwise the tune play request will be sent the whole time that minutes == 0 (which is check many times per second).
 
Comment out the code block (all the if statements) that play your tunes. Does the clock still stop? If the clock does not stop then the calls to mtDFPlayer.play() are the problem. 

I would move the check to play the tune only when the clock is updated because the second has passed. There is no need to check at other times. I would also point out that yo can simplify your code a lot if you make sure that tune number N is played at hour number N:
```
  if (millis() - lastTime >= 1000)
  {
    lastTime += 1000;
    getTime(szTime, flasher);
    flasher = !flasher;
    P.displayReset();
    if (RTC.h == 6 && RTC.m == 35 && RTC.s == 0)  // 6:35:00 Uhr Wecken
        myDFPlayer.play(25);
    else if (RTC.m == 0 && RTC.s == 0)
        myDFPlayer.play(RTC.h);
   }
```

8266er, 2024-04-02:
Hello Marco.

Würdest Du mir dabei auch unter die Arme greifen?

Thanks for your help and patience ;-)

So far it has been running smoothly.

But now I am looking for a solution,

the time information == 19, == 5, == 0 in line 174 (alarm time) as variables via Bluetooth Serial send, without every time upload a new alarm time via IDE.

Would you also help me with this?

Best regards 

Paul

##The current code
````

//Vorlage MD_Parola_Zone_TimeMsg
// Use the DS1307 clock module
#define USE_DS1307 1  // 0

// Header file includes
#include <DFRobotDFPlayerMini.h>
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <MD_DS1307.h>
#include <Wire.h>
#include <SPI.h>

//#include "Font_Data.h"

// Hardware deffinition MAX7219
#define HARDWARE_TYPE MD_MAX72XX::ICSTATION_HW
//#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
//#define HARDWARE_TYPE MD_MAX72XX::GENERIC_HW
//#define HARDWARE_TYPE MD_MAX72XX::PAROLA_HW

#define MAX_DEVICES 4

// for NODE MCU
#define CLK_PIN   14  // D5
#define DATA_PIN  13  // D7
#define CS_PIN    15  // D8

// CONNECTIONS DS1307:
// DS3231 SDA --> SDA
// DS3231 SCL --> SCL
// DS3231 VCC --> 3.3v or 5v
// DS3231 GND --> GND

#define myRTC RTC ///< Library created instance of the RTC class

// DF Player Mini
// D4 ist RX - ESP8266 zu TX - DFPlayer
// D6 ist TX - ESP8266 über 1 kOhm Rvor zu RX - DFPlayer
//SoftwareSerial mySerial(D6, D4);
SoftwareSerial mySerial1(D4, D6);

DFRobotDFPlayerMini myDFPlayer;  // DF Player definieren
/*
  Verzeichnis auf SD Karte muss „MP3“ benannt werden!
  Bis zu 3000 Dateien mit Namensschema 0001.mp3 (bzw. wav), 0002.mp3, ….. 3000.mp3 möglich.
  Alle Dateien sind direkt ansprechbar und keine Probleme mit der Reihenfolge.
*/
[... 179 lines cut ...]
      myDFPlayer.play(7);

    if ( RTC.h == 5 && RTC.m == 0 && RTC.s == 0)   // 05 Uhr
      myDFPlayer.play(6);

    if ( RTC.h == 4 && RTC.m == 0 && RTC.s == 0)   // 04 Uhr
      myDFPlayer.play(5);

    if ( RTC.h == 3 && RTC.m == 0 && RTC.s == 0)   // 03 Uhr
      myDFPlayer.play(4);

    if ( RTC.h == 2 && RTC.m == 0 && RTC.s == 0)   // 02 Uhr
      myDFPlayer.play(3);

    if ( RTC.h == 1 && RTC.m == 0 && RTC.s == 0)   // 01 Uhr
      myDFPlayer.play(2);

    if ( RTC.h == 0 && RTC.m == 0 && RTC.s == 0)   // 00 Uhr
      myDFPlayer.play(1);
    //+++
  }
}

````
[Max7219_RTC_MP3Clock.zip](https://github.com/MajicDesigns/MD_Parola/files/14842421/Max7219_RTC_MP3Clock.zip)

MajicDesigns, 2024-04-02:
Good to know that things are now working more smoothly.

I think this additional functionality is beyond just helping you with the functions of the MD_Parola library. There are many examples on the internet for doing this Bluetooth comms and there is also MD_Parola examples that use Bluetooth to set the display message, so you can look at that. Also, there is the Arduino forum that has many people willing to help for this kind of general problem.

FAK142, 2024-05-01:
Can i use this library for P10 LED Display Panel Module - 32x16 ?

MajicDesigns, 2024-05-01:
Please do not hijack threads for unrelated issues, especially when the original issue is already closed. In future please open a new enquiry.

I don't know what a P10 matrix is. The library will only work with LEDs controlled by the MAX7219 or MAX 7221 IC.

closed 2024-06-20 by MajicDesigns