MajicDesigns/MD_Parola #128: Can't Grow Down A Font Character as One Font When Using 2 FC16 Modules

Raynman77, 2024-04-03:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
I am trying to understand if is even possible to delay the start of a second zone until the 1st zone has completed it's animation cycle.

## Your Environment
** Library Version: 3.7.2 ** 
** Arduino IDE version: 2.3.2 ** 
** Host OS and Version: Windows 10 Pro ** 
** CPU Hardware model/type: Wemos D1 Mini / 2 MD_MAX7219 FC16_HW modules stacked upper zone and lower zone respectively ** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
The provided double height font to animate with a grow down exit or enter effect that starts from the top of the display and finishes at the bottom (grow down starting at row 0 to row 15 ).

## Actual Behaviour
The provided double height font grows down in both zones synchronously similar to a blinds effect.

## Code Demonstrating the Issue
````
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Font_Data.h"

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_ZONES 2
#define ZONE_SIZE 4
#define MAX_DEVICES (MAX_ZONES * ZONE_SIZE)
#define ZONE_UPPER  0
#define ZONE_LOWER  1

#define CLK_PIN   D5
#define DATA_PIN  D7
#define CS_PIN    D2

MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

void setup() {
  P.begin(MAX_ZONES);
  P.setIntensity(2);
  P.setZone(ZONE_LOWER, 0, ZONE_SIZE - 1);
  P.setFont(ZONE_LOWER, BigFontLower);
  P.setZone(ZONE_UPPER, ZONE_SIZE, MAX_DEVICES-1);
  P.setFont(ZONE_UPPER, BigFontUpper);
  P.setCharSpacing(P.getCharSpacing() * 1);
}

void loop() {
  if(P.displayAnimate()) {
    for(uint8_t i=0; i<MAX_ZONES; i++) {
      if(P.getZoneStatus(i)) {
        P.displayZoneText(i, "^", PA_CENTER, 100, 0, PA_PRINT, PA_GROW_DOWN);
      }
    }
  }
}

````

This the custom font
[image]

MajicDesigns, 2024-04-03:
By design library will not synchronize the top/bottom of this animation.

The double height display is a compromise as it is 2 independent displays (zones) that run independently at the same time but not explicitly synchronized exceptthat all zones will use the same time base for animation.

In this situation you could grow down the top zone and when that animation is completed you can start growing down the bottom zone. However this must be synchronized in your code. Obviously the reverse would occur for grow up animation.

closed 2024-04-03 by MajicDesigns

Raynman77, 2024-04-04:
@MajicDesigns Do you perhaps have a really brief example of how to achieve the upper zone growing down first and then after it completes, proceed to grow down the lower zone. I have been beating my head against the wall trying to do this and have not had any success. I've read a lot through the library and have an idea on how to do this but every idea has not yielded this desired effect.

MajicDesigns, 2024-04-04:
Start the animation for the top zone, no animation on the bottom zone.
Run the animation with displayAnimate() and monitor for completion on the top zone only with getZone Status().
Start the animation for the bottom zone, no animation on the top zone.
Run the animation with displayAnimate() and monitor for completion on the bottom zone only with getZone Status().

Raynman77, 2024-04-04:
I just keep ending up with this 
[image]

Raynman77, 2024-04-04:
Ok...I will give it a go Marco, thank you for your suggestiion

Raynman77, 2024-04-04:
Like this?
````
void setup() {
  P.begin(MAX_ZONES);
  P.setIntensity(2);
  P.setZone(ZONE_LOWER, 0, ZONE_SIZE - 1);
  P.setFont(ZONE_LOWER, BigFontLower);
  P.setZone(ZONE_UPPER, ZONE_SIZE, MAX_DEVICES-1);
  P.setFont(ZONE_UPPER, BigFontUpper);
  P.setCharSpacing(P.getCharSpacing() * 1);
}

void loop() {
  if(P.displayAnimate()) {
    if(P.getZoneStatus(ZONE_LOWER)) {
      P.displayZoneText(ZONE_UPPER, "^", PA_CENTER, 100, 0, PA_GROW_DOWN, PA_NO_EFFECT);
      P.displayAnimate();
    }
    if(P.getZoneStatus(ZONE_UPPER)) {
      P.displayZoneText(ZONE_LOWER, "^", PA_CENTER, 100, 0, PA_GROW_DOWN, PA_NO_EFFECT);
      P.displayAnimate();
    }
  }
}
````

Raynman77, 2024-04-04:
@MajicDesigns Tried this too and didn't work.
````
void setup() {
  P.begin(MAX_ZONES);
  P.setIntensity(2);
  P.setZone(ZONE_LOWER, 0, ZONE_SIZE - 1);
  P.setFont(ZONE_LOWER, BigFontLower);
  P.setZone(ZONE_UPPER, ZONE_SIZE, MAX_DEVICES-1);
  P.setFont(ZONE_UPPER, BigFontUpper);
  P.setCharSpacing(P.getCharSpacing() * 1);
}

void loop() {
  static uint8_t curZone = 0;
  P.displayZoneText(ZONE_UPPER, "^", PA_CENTER, 25, 1000, PA_GROW_DOWN, PA_NO_EFFECT);
  while (!P.getZoneStatus(curZone)) {
    P.displayAnimate();
  }
  curZone = ++curZone % MAX_ZONES;
}
````

Raynman77, 2024-04-04:
Ok, so found where my mistake was and corrected it and this is now working. Thank you Marco for the quick responses. 
````
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Font_Data.h"

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_ZONES 2
#define ZONE_SIZE 4
#define MAX_DEVICES (MAX_ZONES * ZONE_SIZE)
#define ZONE_UPPER  0
#define ZONE_LOWER  1

#define CLK_PIN   D5
#define DATA_PIN  D7
#define CS_PIN    D2

MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

void setup() {
  P.begin(MAX_ZONES);
  P.setIntensity(2);
  P.setZone(ZONE_LOWER, 0, ZONE_SIZE - 1);
  P.setFont(ZONE_LOWER, BigFontLower);
  P.setZone(ZONE_UPPER, ZONE_SIZE, MAX_DEVICES-1);
  P.setFont(ZONE_UPPER, BigFontUpper);
  P.setCharSpacing(P.getCharSpacing() * 1);
}

void loop() {
  static uint8_t curZone = 0;
  P.displayZoneText(curZone, "^", PA_CENTER, 75, 0, PA_GROW_DOWN, PA_NO_EFFECT);
  while (!P.getZoneStatus(curZone))
    P.displayAnimate();
    if(curZone == 1) {
      P.displayClear(curZone);
    }
  curZone = ++curZone % MAX_ZONES;
}

````
[image]