MajicDesigns/MD_Parola #130: " 'MD_Parola.h' not found"

tuck9117, 2024-07-28:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
[ ] Have you checked this has not already been submitted and/or resolved?
[ ] If you are requesting help a better choice may be the [Arduino forum](http://forum.arduino.cc/)

## Subject of the issue
Describe your issue here.
When compiling I receive an error for that " 'MD_Parola.h' not found" after it is already included at the top of the program

## Your Environment
**Library Version:** 
**Arduino IDE version:** 
**Host OS and Version:** 
**CPU Hardware model/type:** 

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

## Expected Behaviour
Explain what should happen.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your compilable code code here (no code snippets).
````

MajicDesigns, 2024-07-29:
Did you install the library? You can use the IDE library manager.

tuck9117, 2024-07-29:
Yes I installed the library through the ide, so far I've ran into this
issue with this library as well as the Adafruit_GFX.h library as well but
when I check the files the .h for both are right there.

MajicDesigns, 2024-07-29:
I have seen this type of error when the ibrary has been installed in more than one place. Is this a possible problem? Use the Windows Search to see if the file is reported more than once.

Alternatively you can also try reinstalling the IDE.

tuck9117, 2024-07-30:
Unfortunately I've already tried reinstalling. But I will do a search to
see if it appears in multiple places

closed 2024-09-26 by MajicDesigns