MajicDesigns/MD_Parola #132: issue on Parola_Scrolling_Vertical.ino

dl7ata, 2024-12-25:
Compiler error in statement
 `for (uint16_t i = 0, uint16_t j = len-1; i<j; i++, j--)`
When changing it into 
` for (uint16_t i = 0; uint16_t j = len-1; i<j, i++, j--)` 
compiling without errors works, but ESP32 crashes with 

`[Parola Scrolling Display]
START reverse msg, länge: 25
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4`

## My Environment
**Library Version: _    
majicdesigns/MD_Parola@^3.7.3
 majicdesigns/MD_MAX72XX@^3.5.1_

_Visual Studio 1.96.2
NAME="Linux Mint"
VERSION="22 (Wilma)"
ID=linuxmint
ID_LIKE="ubuntu debian"
PRETTY_NAME="Linux Mint 22"
VERSION_ID="22"
on DELL Precision 3590 (16core, 32gB)_  

````
#include <MD_Parola.h>
#include <MD_MAX72xx.h>
#include <SPI.h>
#include "Parola_Fonts_data.h"
#include <Arduino.h>

// Turn on debug statements to the serial output
#define DEBUG 1

#if DEBUG
#define PRINT(s, x) { Serial.print(F(s)); Serial.print(x); }
#define PRINTS(x) Serial.print(F(x))
#define PRINTX(x) Serial.println(x, HEX)
#else
#define PRINT(s, x)
#define PRINTS(x)
#define PRINTX(x)
#endif

// Define the number of devices we have in the chain and the hardware interface
// NOTE: These pin numbers will probably not work with your hardware and may
// need to be adapted
#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define MAX_DEVICES 8
#define CS_PIN    15

// HARDWARE SPI
MD_Parola P = MD_Parola(HARDWARE_TYPE, CS_PIN, MAX_DEVICES);

// Scrolling parameters
uint8_t scrollSpeed = 25;    // default frame delay value
textEffect_t scrollEffect = PA_SCROLL_LEFT;
textPosition_t scrollAlign = PA_CENTER;
uint16_t scrollPause = 2000; // in milliseconds

// Global message buffers shared by Serial and Scrolling functions
#define	BUF_SIZE	75
char curMessage[BUF_SIZE] = { "" };
char newMessage[BUF_SIZE] = { "Hello! Enter new message?" };
bool newMessageAvailable = true;

void readSerial(void)
{
  static char *cp = newMessage;

  while (Serial.available())
  {
    *cp = (char)Serial.read();
    if ((*cp == '\n') || (cp - newMessage >= BUF_SIZE-2)) // end of message character or full buffer
    {
      *cp = '\0'; // end the string
      // restart the index for next filling spree and flag we have a message waiting
      cp = newMessage;
      newMessageAvailable = true;
    }
    else  // move char pointer to next position
      cp++;
  }
}

void reverse(char *str)
{
  uint16_t len = strlen(str);
  Serial.print("START reverse msg, länge: ");
  Serial.println(len);
  
  for (uint16_t i = 0; uint16_t j = len-1; i<j, i++, j--)
   {
    char a = str[i];
    str[i] = str[j];
    str[j] = a;
    //Serial.println(str[j]);
  }
}

void setup(void)
{
  Serial.begin(115200);
  Serial.println("[Parola Scrolling Display]");

  P.begin();
  P.setFont(_fontVertical);
  P.displayText(curMessage, scrollAlign, scrollSpeed, scrollPause, scrollEffect, scrollEffect);
}

void loop(void)
{
  if (P.displayAnimate())
  {
    if (newMessageAvailable)
    {

      reverse(newMessage);   // reverse the string for proper display order
      strcpy(curMessage, newMessage);
      newMessageAvailable = false;
    }
    P.displayReset();
  }

  readSerial();
}````

MajicDesigns, 2024-12-26:
Please check the length of the string you are trying to convert. If it is a zero length string the loop cold overwrite memory.

dl7ata, 2024-12-27:
Thanks Marco, will do.
And what about the compiler error, is syntax in your scripting ok?

dl7ata, 2024-12-27:
I got it. I changed the function into:
` void reverse(char *str)
{
  Serial.print(str);
  uint16_t len = strlen(str);
  uint16_t i = 0;
  uint16_t j = len-1;

  for (int16_t i = 0; i<=j; i++)
  {
    char a = str[i];
    str[i] = str[j];
    str[j] = a;
    j--;
  }
}`

closed 2024-12-27 by dl7ata

MajicDesigns, 2024-12-28:
The syntax for the for(;;) loop is correct - works for other C/C++ compilers out there and conforms to the standard. What the compiler is trying to change it to is incorrect.

You did the right thing and the rewritten loop is equivalent but obviously better for your situation.