MajicDesigns/MD_Parola #133: Possible bug when using zones and zones==2

waltje, 2025-09-14:
Hello:

I have been looking into a problem reported to me by a friend.  He uses 2 sets ("lines") of dual 4-module FC16' 
(so,8 FC16's per line, so we have a 64x16 display) in a 'single bus' setup, using 2 zones for upper and lower line.

He then uses the BigFontDemo sketch.

This works, but there is an odd artifact.  I can reproduce it 100%, and no, it has nothing to do with timing or
voltage drops for a change... :)

Problem is, that with ANY setup using zones==2, the "first"  display on the chain (so, the one directyly at the
cable end) is considerably more 'dimmed' than the others, whilst the others are super-bright.

When I set the exact same code to zones==1, all looks well, the pixels are medium bright (what the first 
display is with ==2) and I can read the text just fine (just the lower half for some of it ;-)

So.. this SEEMS to be a problem with the zoning code, as other modes and examples do not show this
behavior.

Any ideas, anyone?  Marco?

MajicDesigns, 2025-09-15:
Odd. The library has been unchanged for some time and this has not been previously reported by others.

A few things to consider:
1. Is your friend using the latest version? There was (many years ago) an issue with the memory allocation with multiple zones but that has been fixed.
2. Is the MCU a 3V or 5V? If 3V is the power supplied separately? (I know what you have written about your tests, but pls answer this).
3. If you change the brightness of the dim or other zones in code, do they change as expected?

waltje, 2025-09-15:
[image]This is what "it" looks like, by the way.

MajicDesigns, 2025-09-16:
The example code as shipped does not set the intensity or invert, these have been added by your friend.

I set up a test and could replicate the issue you reported. 

I think the issue is the order of operations:
````
P.begin(MAX_ZONES);
P.setZone(0, 0, ZONE_SIZE-1);
P.setZone(1, ZONE_SIZE, MAX_DEVICES-1);
P.setIntensity(0);  // <--- Move these to AFTER defining the zones
P.setInvert(false);
````
Generally, for multi-zone displays you need to keep the P.begin() and zone definitions (P.setZone()) together. When you are setting the intensity without the library knowing what the zones are there is still notionally one 'zone' of the first module defined because the variables are initialized to zero for each zone start/end pair (ie, module 0 to 0). So any operations that will need to write a value to the modules will write to just this module. Once the correct module segments are defined the library will operate as expected.